PXDiv/Div-Acer-Manager-Max #1: Laptop type: UNKNOWN

Hoag11, 2025-05-05:
Im using fedora 42 on Acer Predator Triton 300 SE PT314-51s but the laptop type is showing UNKNOWN 
I disabled secured boot and my kernel is 6.14.4
I’m newbie 
Please help 😭😭😭

PXDiv, 2025-05-05:
Please just try to reboot once

The codebase is still in early stages, may contain a lot of bugs
Will need a lot of testing.

PXDiv, 2025-05-05:
I have created a new install script and uploaded a new release, please test it out and tell me if any error comes in the install process

There can be two things wrong here first: 
1) The Drivers didn't get installed so the daemon errors out (very likely)
2) The Daemon Freaked out (i dont think this happened)

Please download and try to install the latest build

Hoag11, 2025-05-05:
here is the log:
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Creating default config at /etc/DAMX_Daemon/config.ini
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Log level set to INFO
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Detected laptop type: UNKNOWN
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Base path: 
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Four-zone keyboard: No
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Available features: 
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Detected features: 
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - PowerSourceDetector initialized
2025-05-06 00:28:47,214 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5
2025-05-06 00:28:47,215 - DAMXDaemon - INFO - Monitoring power source started
2025-05-06 00:28:47,215 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock
2025-05-06 00:28:53,695 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
2025-05-06 00:28:53,706 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
2025-05-06 00:29:12,758 - DAMXDaemon - INFO - Received signal 15, shutting down...
2025-05-06 00:29:13,678 - DAMXDaemon - INFO - Cleaning up resources...
2025-05-06 00:29:13,678 - DAMXDaemon - INFO - Stopping server...
2025-05-06 00:29:13,679 - DAMXDaemon - INFO - Daemon stopped
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Log level set to INFO
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Detected laptop type: UNKNOWN
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Base path: 
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Four-zone keyboard: No
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Available features: 
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Detected features: 
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - PowerSourceDetector initialized
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5
2025-05-06 00:46:37,739 - DAMXDaemon - INFO - Monitoring power source started
2025-05-06 00:46:37,740 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock
2025-05-06 00:46:44,556 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
2025-05-06 00:46:44,567 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
2025-05-06 00:47:08,770 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
2025-05-06 00:47:14,582 - DAMXDaemon - INFO - Received signal 15, shutting down...
2025-05-06 00:47:14,583 - DAMXDaemon - INFO - Received signal 15, shutting down...
2025-05-06 00:47:15,549 - DAMXDaemon - INFO - Cleaning up resources...
2025-05-06 00:47:15,549 - DAMXDaemon - INFO - Stopping server...
2025-05-06 00:47:15,549 - DAMXDaemon - INFO - Daemon stopped
2025-05-06 01:16:52,055 - DAMXDaemon - INFO - Log level set to INFO
2025-05-06 01:16:52,055 - DAMXDaemon - INFO - Detected laptop type: UNKNOWN
2025-05-06 01:16:52,055 - DAMXDaemon - INFO - Base path: 
2025-05-06 01:16:52,055 - DAMXDaemon - INFO - Four-zone keyboard: No
2025-05-06 01:16:52,056 - DAMXDaemon - INFO - Available features: 
2025-05-06 01:16:52,056 - DAMXDaemon - INFO - Detected features: 
2025-05-06 01:16:52,056 - DAMXDaemon - INFO - PowerSourceDetector initialized
2025-05-06 01:16:52,056 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5
2025-05-06 01:16:52,056 - DAMXDaemon - INFO - Monitoring power source started
2025-05-06 01:16:52,056 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock
2025-05-06 01:16:59,950 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
2025-05-06 01:16:59,960 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
2025-05-06 01:18:07,655 - DAMXDaemon - INFO - Received signal 15, shutting down...
2025-05-06 01:18:07,987 - DAMXDaemon - INFO - Cleaning up resources...
2025-05-06 01:18:07,988 - DAMXDaemon - INFO - Stopping server...
2025-05-06 01:18:07,988 - DAMXDaemon - INFO - Daemon stopped
2025-05-06 01:18:40,483 - DAMXDaemon - INFO - Log level set to INFO
2025-05-06 01:18:40,483 - DAMXDaemon - INFO - Detected laptop type: UNKNOWN
2025-05-06 01:18:40,483 - DAMXDaemon - INFO - Base path: 
2025-05-06 01:18:40,484 - DAMXDaemon - INFO - Four-zone keyboard: No
2025-05-06 01:18:40,484 - DAMXDaemon - INFO - Available features: 
2025-05-06 01:18:40,484 - DAMXDaemon - INFO - Detected features: 
2025-05-06 01:18:40,484 - DAMXDaemon - INFO - PowerSourceDetector initialized
2025-05-06 01:18:40,484 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5
2025-05-06 01:18:40,484 - DAMXDaemon - INFO - Monitoring power source started
2025-05-06 01:18:40,484 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock
2025-05-06 01:18:49,256 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
2025-05-06 01:18:49,268 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
2025-05-06 01:19:24,406 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
2025-05-06 01:19:24,415 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
2025-05-06 01:19:50,419 - DAMXDaemon - INFO - Received signal 15, shutting down...
2025-05-06 01:19:50,419 - DAMXDaemon - INFO - Received signal 15, shutting down...
2025-05-06 01:19:51,408 - DAMXDaemon - INFO - Cleaning up resources...
2025-05-06 01:19:51,408 - DAMXDaemon - INFO - Stopping server...
2025-05-06 01:19:51,408 - DAMXDaemon - INFO - Daemon stopped
2025-05-06 01:20:30,107 - DAMXDaemon - INFO - Log level set to INFO
[... cut ...]
2025-05-06 01:20:30,107 - DAMXDaemon - INFO - Four-zone keyboard: No
2025-05-06 01:20:30,108 - DAMXDaemon - INFO - Available features: 
2025-05-06 01:20:30,108 - DAMXDaemon - INFO - Detected features: 
2025-05-06 01:20:30,108 - DAMXDaemon - INFO - PowerSourceDetector initialized
2025-05-06 01:20:30,108 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5
2025-05-06 01:20:30,108 - DAMXDaemon - INFO - Monitoring power source started
2025-05-06 01:20:30,108 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock

PXDiv, 2025-05-05:
Yeah the logs indicate that there is no driver to read from, I am sorry for this

Please try the newest release and tell me if it installs properly 
And please show me the error you might face during the install I guess the driver didn't get complied and Installed

Hoag11, 2025-05-05:
Here is my out put when reinstall the newest release:
Starting reinstallation/update...
Uninstalling DAMX Suite...
Stopping DAMX-Daemon service...
Disabling DAMX-Daemon service...
Removed '/etc/systemd/system/multi-user.target.wants/damx-daemon.service'.
Removing installed files...
Uninstalling drivers...
Removing current user from linuwu_sense group if exists...
Removing user hoang from group linuwu_sense
Uninstalled linuwu_sense and cleaned up related configuration.
DAMX Suite uninstalled successfully!
Installing Linuwu-Sense drivers...
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
CLEAN Module.symvers
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
The kernel was built by: gcc (GCC) 15.0.1 20250228 (Red Hat 15.0.1-0)
You are using: gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)
CC [M] src/linuwu_sense.o
MODPOST Module.symvers
CC [M] src/linuwu_sense.mod.o
CC [M] .module-common.o
LD [M] src/linuwu_sense.ko
BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
The kernel was built by: gcc (GCC) 15.0.1 20250228 (Red Hat 15.0.1-0)
You are using: gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
sudo install -d /lib/modules/6.14.0-63.fc42.x86_64/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.14.0-63.fc42.x86_64/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
Created symlink '/etc/systemd/system/multi-user.target.wants/linuwu_sense.service' → '/etc/systemd/system/linuwu_sense.service'.
Setting up group and permissions...
Detected user: hoang
sudo usermod -aG linuwu_sense hoang
Setting permissions via tmpfiles...
Warning: Could not detect predator_sense or nitro_sense in sysfs.
Module linuwu_sense installed and configured to load at boot.
Linuwu-Sense drivers installed successfully!
Installing DAMX-Daemon...
Created symlink '/etc/systemd/system/multi-user.target.wants/damx-daemon.service' → '/etc/systemd/system/damx-daemon.service'.
DAMX-Daemon installed and service started!
Installing DAMX-GUI...
DAMX-GUI installed successfully!
DAMX Suite installation completed successfully!
You can now run the GUI using the DAMX command or from your application launcher.

PXDiv, 2025-05-05:
Oh, on looking at the logs you provided (we'll keep it here as the drivers did get installed and)

It seems like the project is working fine but the drivers don't seem to get what laptop it is. Maybe the linuwu sense drivers don't support it yet?

Since the linuwu sense driver is a different project I would advise you to open an issue to that project and post your device details and the installation log you posted here so your device is added to the compatibility.

The linuwu sense driver project's link is in the readme.

closed 2025-05-05 by Hoag11