PXDiv/Div-Acer-Manager-Max #2: Adding an option to not install Linuwu Drivers when you already have it installed

BenDo85, 2025-05-05:
Hi,
I already had Linuwu module installed on my acer and I just wanted to try out your GUI.
It would be great to add more modularity when asking for installing DAMX only or with Linuwu-Sense too.

Here's an example on how I think it would prompt:

```
┌────────────────────────────────────── - -  -   -
│ DAMX (Div Acer Manager Max) Setup     
│ Version: 0.4.6              
└────────────────────────────────────── - -  -   -

Please select an option:

  1) Install Linuwu-Sense Driver + DAMX
  2) Install DAMX
  3) Uninstall Linuwu-Sense Driver + DAMX
  4) Uninstall DAMX
  5) Reinstall/Update Linuwu-Sense Driver + DAMX
  6) Reinstall/Update DAMX
  7) Exit

Enter your choice [1-7]:
```

Or something like when you select  `1) Install DAMX`
You get prompted:
```
  1) Install Linuwu-Sense Driver + DAMX
  2) Install DAMX
  3) Install Linuwu-Sense Driver
  4) Exit

Enter your choice [1-4]:
```

Have a nice day !

PXDiv, 2025-05-05:
Hi @BenDo85 
Actually I thought of this project as a complete suite with linuwu sense included, so it makes more sense to install everything altogether, as you might have the drivers already but others might not have them. and then also it makes updating easier by removing the driver altogether if a newer version arrives. Furthermore Uninstalling uninstalls all of the files it installed including linuwu-sense. It makes the install process less confusing for new people coming to linux as a whole package. So if you dont want the GUI you just get the linuwu sense from github or if you want the whole thing you get DAMX.

and even if you have it already installed, there must be no problem in going ahead and installing the project, i think it'll work regardless it re installs linuwu sense.
But as you said i made a different install script altogether which has the option to not install the drivers.
Please update me if anything wrong with the install or the DAMX daemon/gui

BenDo85, 2025-05-05:
Hi again,
It's just as of now your GUI doesn't have all the feature that Linuwu driver allow us to use, for example keyboard zone color.
So to compensate for this, I already had Linuwu-Sense-GUI (https://github.com/kumarvivek1752/Linuwu-Sense-GUI) which is why I didn't want to reinstall the driver.
But it still worked for me after some cleaning and a reboot, so I guess it's fine!

closed 2025-05-05 by BenDo85

PXDiv, 2025-05-05:
Yeah it should work that way too.
More features are on the way including the keyboard lighting😉 
Actually, I don't have a laptop that supports the keyboard lighting feature, so i don't have any way to test the feature so i guess the rollout will be a little slow.
And i still have a little more fixes to do.

I have quite some questions to ask you actually :)
How was the GUI and all the other features? 
Did you like it? 
What is your feedback on the project?
Did the thermal Profiles work? 
What is your laptop model? What Distro did you use

BenDo85, 2025-05-05:
I’d be glad to provide feedback once you add keyboard-lighting configuration.

The GUI is really neat! All the thermal features work as expected: I can see the GPU, CPU, and fan speeds change when switching between modes. On battery power, it displays the new thermal profiles (Eco and Balanced). 

One improvement I’d suggest is allowing users to switch the battery thermal profile while on AC power (and vice versa) without having to plug or unplug the cable. It would be more convenient to change presets on the fly. 

Another personal recommendation is to let users toggle Turbo Boost on or off within the Power & Performance section, so everything is centralized. This is how PredatorBlue handles it (see the repo [here](https://github.com/bluebottlewize/predator-blue)). 

Here's how my system settings tab looks
[image]

For reference, my laptop is an Acer Predator PHN16-71, and I’m running Arch Linux.

BenDo85, 2025-05-05:
> The thing is, linuwu sense or more specifically acer laptops, do not allow to switch to modes like quiet, performance and turbo on battery but setting Eco mode while plugged is allowed. You can actually do this by going to developer mode and changing it to eco while plugged in. And you can also test weather its changing to the other higher profiles while on battery to help me include exceptions and allow those buttons to be shown on your laptops on battery.

So when switching to developer mode and changing to eco on AC and battery, It doesn't keep the thermal profile when I unplug the cable and roll back to balance mode.

> I am sorry, but I dont actually understand what TurboBoost is as i do not have the laptop that supports it. Is it the LCD Override setting?

Your computer should have it as it is a commonplace feature for CPU nowaday. [Wiki for Intel Turbo Boost](https://en.wikipedia.org/wiki/Intel_Turbo_Boost)
On Linux, you can enable or disable it by editing the boolean of this file : /sys/devices/system/cpu/intel_pstate/no_turbo

Here how it looks on PredatorBlue:
[image]

PXDiv, 2025-05-06:
Hi @BenDo85,

> Your computer should have it as it is a commonplace feature for CPU nowaday. [Wiki for Intel Turbo Boost](https://en.wikipedia.org/wiki/Intel_Turbo_Boost)
> On Linux, you can enable or disable it by editing the boolean of this file : /sys/devices/system/cpu/intel_pstate/no_turbo

Oh, so its like Intel Turbo Boost Rebranded to include both Intel and AMD Turbo Boost variants as acer makes both of the variants available to avoid confusion they named it under a single umbrella of Turbo Boost.
Ill think of adding the Feature in future, as my current goal is to implement all the features available in nitro/predator sense and what Linuwu-sense provides. Since Linuwu sense dosent support changing it directly, ill refrain from doing it right now as ill have to do it for both intel and amd myself and i am not trying to do that.
and for hyper-threading, I dont think anyone needs to tinker with that, if they need they can find some tutorials easily. They both really show up as advanced features everyone should not mess with

PXDiv, 2025-05-06:
I've Implemented the Keyboard Effects and Lighting in my newest pre-release, Please test if its working (i don't think the light effects will). 
Can I contact you somewhere if you are interested to test out all the features, i don't have all the features available to me and cannot test it myself.

BenDo85, 2025-05-06:
My discord username is [here](https://privnote.com/JIGRU7Do#fwUyjIc2F)
I'll test the light effects later today (around 8 pm UTC+2)