PXDiv/Div-Acer-Manager-Max #3: The Daemon does not detect if the driver is installed or not, just returns Unknown device type

PXDiv, 2025-05-05:
This makes finding and fixing what went wrong harder

Hoag11, 2025-05-05:
Here is my out put when reinstall the newest release:
Starting reinstallation/update...
Uninstalling DAMX Suite...
Stopping DAMX-Daemon service...
Disabling DAMX-Daemon service...
Removed '/etc/systemd/system/multi-user.target.wants/damx-daemon.service'.
Removing installed files...
Uninstalling drivers...
Removing current user from linuwu_sense group if exists...
Removing user hoang from group linuwu_sense
Uninstalled linuwu_sense and cleaned up related configuration.
DAMX Suite uninstalled successfully!
Installing Linuwu-Sense drivers...
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
  CLEAN   Module.symvers
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 15.0.1 20250228 (Red Hat 15.0.1-0)
  You are using:           gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 15.0.1 20250228 (Red Hat 15.0.1-0)
  You are using:           gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
sudo install -d /lib/modules/6.14.0-63.fc42.x86_64/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.14.0-63.fc42.x86_64/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
Created symlink '/etc/systemd/system/multi-user.target.wants/linuwu_sense.service' → '/etc/systemd/system/linuwu_sense.service'.
Setting up group and permissions...
Detected user: hoang
sudo usermod -aG linuwu_sense hoang
Setting permissions via tmpfiles...
Warning: Could not detect predator_sense or nitro_sense in sysfs.
Module linuwu_sense installed and configured to load at boot.
Linuwu-Sense drivers installed successfully!
Installing DAMX-Daemon...
Created symlink '/etc/systemd/system/multi-user.target.wants/damx-daemon.service' → '/etc/systemd/system/damx-daemon.service'.
DAMX-Daemon installed and service started!
Installing DAMX-GUI...
DAMX-GUI installed successfully!
DAMX Suite installation completed successfully!
You can now run the GUI using the DAMX command or from your application launcher.

friday06, 2025-05-05:
i see what do we have here, You sure you've kernel-devel installed ?

Hoag11, 2025-05-06:
i have kernel-devel installed already, in the lastest release it show that `linuwu_sense` could not detect predator_sense or nitro_sense in sysfs.

PXDiv, 2025-05-10:
This issue is fixed and will be addressed in next major release (coming soon)

PXDiv, 2025-05-11:
This issue has been fixed in the newest release

closed 2025-05-11 by PXDiv