PXDiv/Div-Acer-Manager-Max #5: Question about power profiles

Melk047, 2025-05-09:
My Desktop Environment (KDE plasma) has power profiles built in, which i always have on performance mode. Do the DAMX power profiles override this or should i keep both on performance?

[image]

PXDiv, 2025-05-10:
No, the DAMX does not interfere with the built-in profiles and they are separate.
DAMX changes the performance profiles on a whole system level as provided and tuned by acer with WMI methods and does not do things like reduce brightness for more battery and OS stuff like that.
The built-in profiles do everything possible from the Linux side to increase or decrease the performance including brightness reducing (sometime they get the right drivers and do the right stuff on laptops)

Melk047, 2025-05-10:
> No, the DAMX mostly does not interfere with the built-in profiles and they are separate.
> DAMX changes the performance profiles on a whole system level as provided and tuned by acer with WMI methods and does not do things like reduce brightness for more battery and OS stuff like that.
> The built-in profiles do everything possible from the Linux side to increase or decrease the performance including brightness reducing (sometime they get the right drivers and do the right stuff on laptops)

so i would probably be better off leavimg built in on performance and just using the DAMX ones?

PXDiv, 2025-05-10:
Yeah, using the built-in profiles i can see that on Power Saving, Linux does reduce the performance a lot (it lags) so putting on performance mode may also use more power and generate heat but should give some more performance.

At last it up-to you what mode you wanna use, anything is fine

Melk047, 2025-05-10:
> Yeah, using the built-in profiles i can see that on Power Saving, Linux does reduce the performance a lot (it lags) so putting on performance mode may also use more power and generate heat but should give some more performance. If its worth it you can put it on performance mode if you like. I generally put it in Balanced mode.
> 
> At last it up-to you what mode you wanna use, anything is fine

okay, thanks for the clarification

PXDiv, 2025-05-10:
No problem, if its clear Should I close the Issue now?

Melk047, 2025-05-10:
> No problem, if its clear Should I close the Issue now?

yeah go ahead

closed 2025-05-10 by PXDiv