PXDiv/Div-Acer-Manager-Max #6: Linuwu-sense fails to install

rafalmozzalera45, 2025-05-10:
here is my reinstall output:

Starting reinstallation/update...
Uninstalling DAMX Suite...
Stopping DAMX-Daemon service...
Disabling DAMX-Daemon service...
Removed "/etc/systemd/system/multi-user.target.wants/damx-daemon.service".
Removing installed files...
Uninstalling drivers...
DAMX Suite uninstalled successfully!
Installing Linuwu-Sense drivers...
make -C /lib/modules/6.8.0-59-generic/build M=/home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-59-generic'
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-59-generic'
make -C /lib/modules/6.8.0-59-generic/build M=/home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-59-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  /home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense/src/linuwu_sense.o
/home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense/src/linuwu_sense.c:35:11: fatal error: linux/unaligned.h: No such file or directory
   35 |  #include <linux/unaligned.h>
      |           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:243: /home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense/src/linuwu_sense.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-59-generic/Makefile:1925: /home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-59-generic'
make: *** [Makefile:12: all] Error 2
make -C /lib/modules/6.8.0-59-generic/build M=/home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-59-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  /home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense/src/linuwu_sense.o
/home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense/src/linuwu_sense.c:35:11: fatal error: linux/unaligned.h: No such file or directory
   35 |  #include <linux/unaligned.h>
      |           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:243: /home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense/src/linuwu_sense.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-59-generic/Makefile:1925: /home/rafal/Downloads/DAMX-0.4.6b/Linuwu-Sense] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-59-generic'
make: *** [Makefile:12: all] Error 2
Error: Failed to install Linuwu-Sense drivers
Installing DAMX-Daemon...
Created symlink /etc/systemd/system/multi-user.target.wants/damx-daemon.service → /etc/systemd/system/damx-daemon.service.
DAMX-Daemon installed and service started!
Installing DAMX-GUI...

and here are my logs:

[DAMX_Daemon_Log.log](https://github.com/user-attachments/files/20141012/DAMX_Daemon_Log.log)

My laptop is an Acer Nitro V15 (Nitro ANV15-51 V1.14) if that helps

PXDiv, 2025-05-10:
This issue is known with Ubuntu 24.04 and its kernel 6.11 (I had this issue too)

Using a system with kernel with 6.12 or higher will fix this issue like ubuntu 25.05 (the version this project was made in, the same laptop as you too). I recommend updating to that from my side.

And since this is actually a issue related to the linuwu sense driver i recommend opening a issue on that project ([here](https://github.com/0x7375646F/Linuwu-Sense)) and uploading your logs there too, that might get it fixed it too. But since i don't acutally maintain the driver and don't really know whats wrong with it, i cannot solve the issue myself with the driver but only can recommend updating to a newer kernel or a newer ubuntu version

rafalmozzalera45, 2025-05-10:
> This issue is known with Ubuntu 24.04 and its kernel 6.11 (I had this issue too)

I forgot to add, but i am running Linux Mint 22.1 x86_64. But nontheless the issue stands so ill do as you said and make an issue on the linuwu project page.

PXDiv, 2025-05-11:
Yeah, anything under the kernel 6.12 won't work properly with the drivers.
If possible try to update the kernel for distro