PXDiv/Div-Acer-Manager-Max #7: Temperature values are not displayed correctly

MematiBas42, 2025-05-15:
[image]

> 2025-05-15 13:29:45,606 - DAMXDaemon - INFO - PowerSourceDetector initialized
> 2025-05-15 13:29:45,606 - DAMXDaemon - INFO - PowerSourceDetector initialized
> 2025-05-15 13:29:45,606 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.6
> 2025-05-15 13:29:45,606 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5
> 2025-05-15 13:29:45,606 - DAMXDaemon - INFO - Monitoring power source started
> 2025-05-15 13:29:45,606 - DAMXDaemon - INFO - Monitoring power source started
> 2025-05-15 13:29:45,607 - DAMXDaemon - ERROR - Failed to start server: [Errno 98] Address already in use
> 2025-05-15 13:29:45,607 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock
> 2025-05-15 13:29:45,607 - DAMXDaemon - INFO - Cleaning up resources...
> 2025-05-15 13:29:45,607 - DAMXDaemon - INFO - Stopping server...
> 2025-05-15 13:29:45,607 - DAMXDaemon - INFO - Daemon stopped
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - Log level set to INFO
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - linuwu_sense module found. Proceeding with initialization.
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - Detected laptop type: NITRO
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - Base path: /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - Four-zone keyboard: No
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - Available features: thermal_profile, battery_limiter, usb_charging, fan_speed, battery_calibration
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - Detected features: battery_calibration, battery_limiter, fan_speed, thermal_profile, usb_charging
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - PowerSourceDetector initialized
> 2025-05-15 16:14:32,125 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.6
> 2025-05-15 16:14:32,130 - DAMXDaemon - INFO - Switched to AC power
> 2025-05-15 16:14:32,130 - DAMXDaemon - INFO - Monitoring power source started
> 2025-05-15 16:14:32,131 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock
> 2025-05-15 16:14:36,943 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
> 2025-05-15 16:14:36,950 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
> 2025-05-15 16:14:41,568 - DAMXDaemon - INFO - Processing command: set_thermal_profile with params: {'profile': 'balanced-performance'}
> 2025-05-15 16:14:42,576 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
> 2025-05-15 16:15:10,526 - DAMXDaemon - INFO - Processing command: set_thermal_profile with params: {'profile': 'quiet'}
> 2025-05-15 16:15:18,314 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
> 2025-05-15 16:15:18,343 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
> 2025-05-15 16:15:18,517 - DAMXDaemon - INFO - Processing command: set_thermal_profile with params: {'profile': 'quiet'}
> 2025-05-15 16:15:18,592 - DAMXDaemon - INFO - Processing command: set_fan_speed with params: {'cpu': 0, 'gpu': 0}
> 2025-05-15 16:15:19,621 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
> 2025-05-15 16:15:24,959 - DAMXDaemon - INFO - Processing command: set_thermal_profile with params: {'profile': 'balanced'}
> 2025-05-15 16:15:25,146 - DAMXDaemon - INFO - Processing command: set_thermal_profile with params: {'profile': 'balanced-performance'}
> 2025-05-15 16:15:29,700 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {}
> 2025-05-15 16:15:29,706 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}
> 2025-05-15 16:15:29,774 - DAMXDaemon - INFO - Processing command: set_thermal_profile with params: {'profile': 'balanced-performance'}
> 2025-05-15 16:15:30,782 - DAMXDaemon - INFO - Processing command: get_all_settings with params: {}

PXDiv, 2025-05-16:
The current implementation is not very accurate as it reads from the 6th or 2nd core of the cpu, I didnt choose the first core as it gets hot pretty quick and does not represent and the actual sensation felt on touching the laptop (like it ramped to 80 on small tasks), the future implementation will be much accurate as it will take all the cores, avg them out, then read the temp.

MematiBas42, 2025-05-20:
[image]

thank you so much i guess the difference is because of read method but it is better now acording to older version

PXDiv, 2025-05-20:
Glad to hear that, 
Can you please provide me with the dashboard temp and inbuilt temp reader value at the same time screenshot?

MematiBas42, 2025-05-20:
Yeah sure

[image]

[image]

PXDiv, 2025-05-20:
Thank you,
seems like the temperature display is working pretty good now. Ill close this issue then

closed 2025-05-20 by PXDiv