PXDiv/Div-Acer-Manager-Max #8: Battery Limiter

Melk047, 2025-05-18:
I was dual booting for the longest time, i recently installed just EOS (Arch based) onto my system without windows, DAMX hasnt really worked in limiting my battery. Acer care worked so well it continued to work in linux which made me think that it worked with DAMX. But it is not limiting the battery whatsoever. 

Model:
Nitro ANV15-51

And i don't know if this is important but here is the kernel im on:
Linux 6.14.6-arch1-1

Melk047, 2025-05-18:
This i assume is just a cosmetic thing but this happens too, unless you made it so when battery limiter is on 80% looks like 100% to the system?

[image]

MematiBas42, 2025-05-18:
Im on  EndeavourOS Linux 6.14.6-zen1-1-zen with Nitro ANV15-51 (V1.26) and i can confirm that battery limiter works for me, probably your setting are conflicting with windows nitro sense app. Try enabling on windows. (That is what i did)

PXDiv, 2025-05-19:
Hmm... so that's an interesting one to be honest.
The battery bar is a visual glitch and has nothing to do with this issue so i referenced it in another issue.

I know for a fact that the battery limiter thing turns off after sometime randomly even though you didn't turn it off (This happens in windows too). But this bug is pretty interesting too me, specifically that it happened when not dual booting.
In theory it should work regardless of whether windows is installed or not as it uses the same WMI methods to do it.
Try reinstalling the app (pressing 4 in the setup) then turning off and on the battery limiter again.
If that didn't work, try this into the terminal:

> echo 1 | sudo tee /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/battery_limiter

and can you please give me the logs? its in the troubleshooting section on how to get them.

Melk047, 2025-05-19:
> Im on  EndeavourOS Linux 6.14.6-zen1-1-zen with Nitro ANV15-51 (V1.26) and i can confirm that battery limiter works for me, probably your settings are conflicting with windows nitro sense app. Try enabling limiter on windows and reboot linux. (That is what i did)

I don't have windows anymore, after i removed windows thats when it stopped working. Thats why i think the DAMX battery limiter doesnt work and that was just the windows one being built into the hardware.

Melk047, 2025-05-19:
> Try reinstalling the app (pressing 4 in the setup) then turning off and on the battery limiter again.
> If that didn't work, try this into the terminal:
> 
> > echo 1 | sudo tee /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/battery_limiter
>  

Okay, I'll try all this when i get home later today

Melk047, 2025-05-19:
> Try reinstalling the app (pressing 4 in the setup) then turning off and on the battery limiter again.
> 

Just an update. that this happened after reinstalling. 

[image]

Melk047, 2025-05-19:
and i don't have any settings. overview is showing (assumably correct) information though. 

[image]

Melk047, 2025-05-19:
And as for the command you asked me to use if reinstalling didn't work.

[image]

And as for the logs 

[DAMX_Daemon_Log.log](https://github.com/user-attachments/files/20319518/DAMX_Daemon_Log.log)

Melk047, 2025-05-20:
So i got it working, reinstalling worked but i had to manually uninstall then reinstall, the automatic reinstall script doesnt install the drivers properly it seems.

closed 2025-05-23 by Melk047