PXDiv/Div-Acer-Manager-Max #9: Please consider adding a socket cleaning mechanism

MematiBas42, 2025-05-18:
_[Unit]
Description=DAMX Daemon Service for Acer laptops
After=network.target_

_[Service]
Type=simple_
**ExecStartPre=/bin/rm -f /var/run/DAMX.sock**
_ExecStart=/opt/DAMX/DAMX-Daemon/DAMX-Daemon
Restart=on-failure
User=root
Group=root_

_[Install]
WantedBy=multi-user.target_

Because sometimes system service fails to start after "socket already in use" error.

<details>
<summary>Logcat Output</summary>
 ➜ sudo journalctl -u DAMX-Daemon.service -b

May 19 01:05:50 nitro-nitroanv1551 systemd[1]: Started DAMX Daemon Service for Acer laptops.

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - Log level set to INFO

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - Detected laptop type: NITRO

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - Base path: /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - Four-zone keyboard: No

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - Available features: fan_speed, battery_calibration, thermal_profile, battery_limiter, u>

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - Detected features: battery_calibration, battery_limiter, fan_speed, thermal_profile, us>

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - PowerSourceDetector initialized

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,439 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,449 - DAMXDaemon - INFO - Switched to AC power

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,450 - DAMXDaemon - INFO - Monitoring power source started

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,450 - DAMXDaemon - ERROR - Failed to start server: [Errno 98] Address already in use

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,450 - DAMXDaemon - INFO - Cleaning up resources...

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,450 - DAMXDaemon - INFO - Stopping server...

May 19 01:05:50 nitro-nitroanv1551 DAMX-Daemon[1054]: 2025-05-19 01:05:50,450 - DAMXDaemon - INFO - Daemon stopped

May 19 01:05:50 nitro-nitroanv1551 systemd[1]: DAMX-Daemon.service: Deactivated successfully.

May 19 01:27:31 nitro-nitroanv1551 systemd[1]: Started DAMX Daemon Service for Acer laptops.

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,114 - DAMXDaemon - INFO - Log level set to INFO

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,115 - DAMXDaemon - INFO - Detected laptop type: NITRO

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,115 - DAMXDaemon - INFO - Base path: /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,115 - DAMXDaemon - INFO - Four-zone keyboard: No

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,115 - DAMXDaemon - INFO - Available features: thermal_profile, fan_speed, usb_charging, battery_limiter, battery>

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,115 - DAMXDaemon - INFO - Detected features: battery_calibration, battery_limiter, fan_speed, thermal_profile, u>

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,116 - DAMXDaemon - INFO - PowerSourceDetector initialized

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,116 - DAMXDaemon - INFO - Starting DAMX-Daemon v0.2.5

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,137 - DAMXDaemon - INFO - Switched to AC power

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,139 - DAMXDaemon - INFO - Monitoring power source started

May 19 01:27:32 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:32,139 - DAMXDaemon - INFO - Server listening on /var/run/DAMX.sock

May 19 01:27:38 nitro-nitroanv1551 DAMX-Daemon[12089]: 2025-05-19 01:27:38,255 - DAMXDaemon - INFO - Processing command: get_supported_features with params: {} 
</details>

PXDiv, 2025-05-19:
Thank you for pointing this out
I'll look into this and more of the stuff that you recommended, thank you for your attentions towards this project.
My exams are going on currently so won't be able to develop anything right now but will push out a massive update to the project when I get free.
Thank you again :)

MematiBas42, 2025-05-19:
Please do not rush, we want the developer of our project to have a good academic career. <3

closed 2025-05-21 by MematiBas42