PXDiv/Div-Acer-Manager-Max #13: I cant do anything with this app

zkhribach, 2025-05-21:
I have an Acer Nitro AN16-41

```bash
Kernel Version : 6.14.6-arch1-1
OS : Archlinux
DE: KDE PLASMA
```

[image]

[image]

[image]

MematiBas42, 2025-05-21:
`Hello sometimes updates become really a headache, i reinstall/update, delete - install, again instal more update but no luck, performance mode selection not appears. At the end i rollbacked to my snapshot, deleted everything related DAMX, installed version 6 and updated carefully by hand and everything worked. I belive uninstall - install (maybe combinations couple of times) should do the trick also.`

This was my comment on another issue so try uninstall - install or uninstall- install - reinstall/update combinations untill fan controls starts working, also you can do 

`sudo systemctl restart DAMX-Daemon.service`

zkhribach, 2025-05-21:
> `sudo systemctl restart DAMX-Daemon.service`
> 
> after uninstall - install or uninstall- install - reinstall/update combinations.

im getting this after doing the combination

```bash
[diva@archlinux DAMX-0.7.2]$ sudo systemctl restart DAMX-Daemon.service
Failed to restart DAMX-Daemon.service: Unit DAMX-Daemon.service not found.
```

zkhribach, 2025-05-21:
ok it was this 
`damx-daemon.service`

but still not fixed

PXDiv, 2025-05-22:
Hello there,
Just to confirm, are you still unable to use the features?
That would mean that the AN16-41 isnt fully supported, as I can see in the logs, the driver is successfully installed so is the daemon and service.
The Driver is found and is reporting that the device just supports thermal profile

> 2025-05-21 12:19:02,237 - DAMXDaemon - INFO - linuwu_sense module found. Proceeding with initialization.
> 2025-05-21 12:19:02,237 - DAMXDaemon - INFO - Detected laptop type: UNKNOWN
> 2025-05-21 12:19:02,237 - DAMXDaemon - INFO - Base path: 
> 2025-05-21 12:19:02,237 - DAMXDaemon - INFO - Four-zone keyboard: No
> 2025-05-21 12:19:02,237 - DAMXDaemon - INFO - Available features: thermal_profile
> 2025-05-21 12:19:02,237 - DAMXDaemon - INFO - Detected features: thermal_profile
> 2025-05-21 12:19:02,237 - DAMXDaemon - INFO - PowerSourceDetector initialized

Here I can see that Linuwu se working as expected, but the laptop is not supported and only thermal profile is detected as a feature but thats not working too.
Please go to the issues section of linuwu sense and create an issue requesting support 
https://github.com/0x7375646F/Linuwu-Sense

zkhribach, 2025-05-22:
@PXDiv i created an issue here and i hope he fixes it
https://github.com/0x7375646F/Linuwu-Sense/issues/20

zkhribach, 2025-05-24:
@PXDiv i tried this guys patch 
https://github.com/0x7375646F/Linuwu-Sense/pull/17
and its working can you merge it

zkhribach, 2025-05-24:
[image]
fans controls are working omg
keyboard is also working