PXDiv/Div-Acer-Manager-Max #14: Unable to install on Linux Mint 22.1 Kernel 6.13.7

AbishkarM211, 2025-05-21:
I've tried to install it on my Acer Nitro 16 AN16-43 but it fails to install drivers.

```
make -C /lib/modules/6.13.7-061307-generic/build M=/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/linux-headers-6.13.7-061307-generic'
make[2]: Entering directory '/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense'
make[2]: Leaving directory '/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/linux-headers-6.13.7-061307-generic'
make -C /lib/modules/6.13.7-061307-generic/build M=/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.13.7-061307-generic'
make[2]: Entering directory '/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-14 (Ubuntu 14.2.0-4ubuntu2) 14.2.0
  You are using:           gcc-14 (Ubuntu 14.2.0-4ubuntu2~24.04) 14.2.0
  CC [M]  src/linuwu_sense.o
src/linuwu_sense.c:2201:22: error: variable ‘acer_predator_v4_platform_profile_ops’ has initializer but incomplete type
 2201 |  static const struct platform_profile_ops acer_predator_v4_platform_profile_ops = {
      |                      ^~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:2202:7: error: ‘const struct platform_profile_ops’ has no member named ‘probe’
 2202 |      .probe = acer_predator_v4_platform_profile_probe,
      |       ^~~~~
src/linuwu_sense.c:2202:15: warning: excess elements in struct initializer
 2202 |      .probe = acer_predator_v4_platform_profile_probe,
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:2202:15: note: (near initialization for ‘acer_predator_v4_platform_profile_ops’)
src/linuwu_sense.c:2203:7: error: ‘const struct platform_profile_ops’ has no member named ‘profile_get’
 2203 |      .profile_get = acer_predator_v4_platform_profile_get,
      |       ^~~~~~~~~~~
src/linuwu_sense.c:2203:21: warning: excess elements in struct initializer
 2203 |      .profile_get = acer_predator_v4_platform_profile_get,
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:2203:21: note: (near initialization for ‘acer_predator_v4_platform_profile_ops’)
src/linuwu_sense.c:2204:7: error: ‘const struct platform_profile_ops’ has no member named ‘profile_set’
 2204 |      .profile_set = acer_predator_v4_platform_profile_set,
      |       ^~~~~~~~~~~
src/linuwu_sense.c:2204:21: warning: excess elements in struct initializer
 2204 |      .profile_set = acer_predator_v4_platform_profile_set,
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:2204:21: note: (near initialization for ‘acer_predator_v4_platform_profile_ops’)
src/linuwu_sense.c: In function ‘acer_platform_profile_setup’:
src/linuwu_sense.c:2210:36: error: implicit declaration of function ‘devm_platform_profile_register’; did you mean ‘platform_profile_register’? [-Wimplicit-function-declaration]
 2210 |          platform_profile_device = devm_platform_profile_register(
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                    platform_profile_register
src/linuwu_sense.c:2210:34: error: assignment to ‘struct device *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 2210 |          platform_profile_device = devm_platform_profile_register(
      |                                  ^
src/linuwu_sense.c: In function ‘acer_thermal_profile_change’:
src/linuwu_sense.c:2295:10: error: too many arguments to function ‘platform_profile_notify’
 2295 |          platform_profile_notify(platform_profile_device);
      |          ^~~~~~~~~~~~~~~~~~~~~~~
In file included from src/linuwu_sense.c:22:
/usr/src/linux-headers-6.13.7-061307-generic/include/linux/platform_profile.h:40:6: note: declared here
   40 | void platform_profile_notify(void);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c: At top level:
src/linuwu_sense.c:2201:43: error: storage size of ‘acer_predator_v4_platform_profile_ops’ isn’t known
 2201 |  static const struct platform_profile_ops acer_predator_v4_platform_profile_ops = {
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-6.13.7-061307-generic/scripts/Makefile.build:196: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.13.7-061307-generic/Makefile:1996: .] Error 2
make[2]: *** [/usr/src/linux-headers-6.13.7-061307-generic/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.13.7-061307-generic'
make: *** [Makefile:12: all] Error 2
make -C /lib/modules/6.13.7-061307-generic/build M=/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.13.7-061307-generic'
make[2]: Entering directory '/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
[... 33 lines cut ...]
 2210 |          platform_profile_device = devm_platform_profile_register(
      |                                  ^
src/linuwu_sense.c: In function ‘acer_thermal_profile_change’:
src/linuwu_sense.c:2295:10: error: too many arguments to function ‘platform_profile_notify’
 2295 |          platform_profile_notify(platform_profile_device);
      |          ^~~~~~~~~~~~~~~~~~~~~~~
In file included from src/linuwu_sense.c:22:
/usr/src/linux-headers-6.13.7-061307-generic/include/linux/platform_profile.h:40:6: note: declared here
   40 | void platform_profile_notify(void);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c: At top level:
src/linuwu_sense.c:2201:43: error: storage size of ‘acer_predator_v4_platform_profile_ops’ isn’t known
 2201 |  static const struct platform_profile_ops acer_predator_v4_platform_profile_ops = {
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-6.13.7-061307-generic/scripts/Makefile.build:196: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.13.7-061307-generic/Makefile:1996: .] Error 2
make[2]: *** [/usr/src/linux-headers-6.13.7-061307-generic/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/marcelune/Downloads/DAMX-0.6.1/Linuwu-Sense'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.13.7-061307-generic'
make: *** [Makefile:12: all] Error 2
Error: Failed to install Linuwu-Sense drivers
Press any key to continue...

```

MematiBas42, 2025-05-21:
Verify Header Files: Ensure that the kernel headers installed on your system match the running kernel version. You can check the installed headers with:

    ls /usr/src/linux-headers-$(uname -r)

If the headers are missing or don't match the kernel version, install them using your distribution's package manager. For Ubuntu-based systems:

sudo apt install linux-headers-$(uname -r)

AbishkarM211, 2025-05-21:
This is the output I get when checking the header files:

`arch  block  certs  crypto  Documentation  drivers  fs  include  init  io_uring  ipc  Kbuild  Kconfig  kernel  lib  Makefile  mm  Module.symvers  net  rust  samples  scripts  security  sound  tools  usr  virt`

Trying to update it reports that the headers are already the latest version.

PXDiv, 2025-05-22:
Its because the differences in kernel i guess
Please post the driver related issue to linuwu sense github page issue with the logs, disto, model info
https://github.com/0x7375646F/Linuwu-Sense

AbishkarM211, 2025-05-25:
@PXDiv  I've just updated to kernel 6.14.8, and after modifying 0x7375646F/Linuwu-Sense#17 to include an16-43, I can happily report everything in working order 😁

closed 2025-05-25 by AbishkarM211