PXDiv/Div-Acer-Manager-Max #17: Has the tendency to just not work

Melk047, 2025-06-02:
Every now and then (almost every system restart) the app doesnt work. Restarting the daemon shows that theres no such thing, and the only thing i can find that works is just reinstalling the app. 

[image]

[image]

[image]

PXDiv, 2025-06-03:
When on Unknown Model:
Please provide me with the result of `sudo journalctl -k | grep -i linuwu`

and try restarting the linuwu drivers several times until it works by 
```
sudo rmmod linuwu-sense
sudo modprobe linuwu-sense
```

and restarting the damx gui everytime

Melk047, 2025-06-05:
It's fixed. Thank you, and thank you for your hard work on this project its very well done.

PXDiv, 2025-06-06:
Thanks for confirming its working, ill be closing this issue here.
I would still like to know whats the distro that you are using with kernel info too

closed 2025-06-06 by PXDiv