PXDiv/Div-Acer-Manager-Max #22: Got THAT issue again.

Melk047, 2025-06-07:
[image]

and here are my specs as asked previously, i know you don't need all of this but its easier than cropping it a bunch more.

[image]

Melk047, 2025-06-07:
[image]

dunno if this is useful but heres your check service status results.

Melk047, 2025-06-08:
Dug thru to my old issue. everything you asked for

[image]

[image]

Eventually, reinstalling did fix it. Still annoying to have to do every couple restarts though

PXDiv, 2025-06-10:
As stated in: https://github.com/PXDiv/Div-Acer-Manager-Max/issues/10#issuecomment-2939756225

To test the fix out (One time solution)
```
sudo modprobe -r linuwu_sense && sudo modprobe linuwu_sense nitro_v4=1
systemctl restart damx-daemon.service
```
Restart the GUI

To permanently set the argument to force nitro model, use:
`sudo echo "options linuwu_sense nitro_v4=1" > /etc/modprobe.d/linuwu-sense.conf`

then reboot the system

kzrob, 2025-06-11:
Hi, I'm also facing this issue where it says my laptop type is unknown. I tried rebooting, but it didn't work. It's detecting my CPU and GPU temperatures, but not my fans.

## Hardware Information:
- **Hardware Model:**                              Acer Predator PT516-51s
- **Memory:**                                      16.0 GiB
- **Processor:**                                   11th Gen Intel® Core™ i7-11800H × 16
- **Graphics:**                                    Intel® UHD Graphics (TGL GT1)
- **Graphics 1:**                                  NVIDIA GeForce RTX™ 3060 Laptop GPU
- **Disk Capacity:**                               512.1 GB

## Software Information:
- **Firmware Version:**                            V1.07
- **OS Name:**                                     Fedora Linux 42 (Workstation Edition)
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               48
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.14.9-300.fc42.x86_64

PXDiv, 2025-06-12:
> Hi, I'm also facing this issue where it says my laptop type is unknown. I tried rebooting, but it didn't work. It's detecting my CPU and GPU temperatures, but not my fans.
>
> ## Hardware Information: [...]

hello @kzrob, The linuwu-sense drivers may not have the support for your model, please go to the linuwu-sense github and post a issue there to get the support for the model (slower), or a better way will be to make the support yourself by including your model in the driver and then merging them with the main drivers like many other users have done.
Keep in mind this will only work if your model uses wmi methods like most modern acer machines (2022+)

PXDiv, 2025-06-12:
@Melk047 Please edit the issue title to include more information about the issue to make it easier to navigate for the other users that may be facing the same issue. 
GitHub also has the feature to reopen the last issue if the solution didn't work.

Melk047, 2025-06-16:
> can i see the installation logs?
> 

if it happens again, this issue is weird as it happens sometimes but other times its fine

Melk047, 2025-06-27:
I just updated, give me a few days just to make sure the issue doesn't come back, i will close the issue when im confident i will not run into it again.

Melk047, 2025-06-30:
closing now.

[image]

[image]

however this new issue exists with this update, it seems to just be cosmetic. restarting the driver fixes it.

closed 2025-06-30 by Melk047

PXDiv, 2025-06-30:
Thanks for confirming that the new drivers work.

The Issue you described has already been solved internally with tag v0.9.0 will be released sometime later.

Thank you again for contributing to DAMX