PXDiv/Div-Acer-Manager-Max #28: Failing to install

XRGC-Digital, 2025-06-16:
Hello! I happen to have a PH16-71 (i9-13900HX, 4080) that isn't installing DAMX right.

Console log:
...
Installing Linuwu-Sense drivers...
make -C /lib/modules/6.14.0-15-generic/build M=/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-15-generic'
make[2]: Entering directory '/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense'
  CLEAN   Module.symvers
make[2]: Leaving directory '/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0-15-generic'
make -C /lib/modules/6.14.0-15-generic/build M=/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-15-generic'
make[2]: Entering directory '/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
  You are using:           gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0-15-generic'
make -C /lib/modules/6.14.0-15-generic/build M=/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-15-generic'
make[2]: Entering directory '/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
  You are using:           gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
make[2]: Leaving directory '/home/lg/Downloads/DAMX-0.7.10/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0-15-generic'
sudo install -d /lib/modules/6.14.0-15-generic/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.14.0-15-generic/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
modprobe: ERROR: could not insert 'linuwu_sense': Key was rejected by service
make: *** [Makefile:45: install] Error 1
Error: Failed to install Linuwu-Sense drivers
Press any key to continue...

On Ubuntu 25 with 6.14 Kernel.

Any advice on how to fix?

Thanks!

PXDiv, 2025-06-21:
Is it fixed now?

XRGC-Digital, 2025-06-21:
> PXDiv left a comment [(PXDiv/Div-Acer-Manager-Max#28)](https://github.com/PXDiv/Div-Acer-Manager-Max/issues/28#issuecomment-2993705800)
>
> Is it fixed now?

Yep, it did! I forgot to mention, got busy at work lol. The lighting ones don’t work for that model yet though right?