PXDiv/Div-Acer-Manager-Max #31: Laptop type not detected

PierreRamez, 2025-06-22:
AN-517-54
Acer Nitro 5 Core i7-11800H, GeForce RTX 3050Ti, 17.3", 144Hz

PierreRamez, 2025-06-30:
It always says "failed to install Linuwu-Sense drivers" also says "Some components failed to install. Please check the errors above.". also I can't find it in the application menu

PXDiv, 2025-06-30:
Please also post all the logs that appeared during installation.

PierreRamez, 2025-07-01:
I turned off the secure boot and the installation process went well. The problem now is that the laptop model is still unknown, and the fan speeds are not read by the program, neither any other program.

PXDiv, 2025-07-13:
Since there is no update to the last reply in the last one week, I'll assume the issue has been fixed.
Closing the issue now, please reopen the issue if you still have any problems.

closed 2025-07-13 by PXDiv