PXDiv/Div-Acer-Manager-Max #39: Linuwu-Sense fails to install after running script.

yourpepsidrinkwastaken, 2025-07-02:
Hello, I'm using CachyOS with linux-cachyos-bore-lto installed and default linux-cachyos installed incase of failure.

Right now, I tried to install DAMX and this is my output : 

make -C /lib/modules/6.15.4-2-cachyos-bore-lto/build M=/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense clean
make[1]: Entering directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make[2]: Entering directory '/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense'
make[2]: Leaving directory '/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make -C /lib/modules/6.15.4-2-cachyos-bore-lto/build M=/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make[2]: Entering directory '/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 20.1.6
  You are using:           gcc (GCC) 15.1.1 20250425
  CC [M]  src/linuwu_sense.o
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
make[4]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/scripts/Makefile.build:212: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/Makefile:2039: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make: *** [Makefile:12: all] Error 2
make -C /lib/modules/6.15.4-2-cachyos-bore-lto/build M=/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make[2]: Entering directory '/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 20.1.6
  You are using:           gcc (GCC) 15.1.1 20250425
  CC [M]  src/linuwu_sense.o
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
make[4]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/scripts/Makefile.build:212: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/Makefile:2039: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/tmp/damx-install-59209/DAMX-0.9.1/Linuwu-Sense'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make: *** [Makefile:12: all] Error 2

I also tried to install Linuwu drivers and reinstall DAMX, but it fails because of the same output.

Melk047, 2025-07-02:
i have this issue too, both local and online install.

Melk047, 2025-07-02:
a simple restart fixed it. (local install)

i downloaded the tar file, extracted, used the script to uninstall damx suite then rebooted, and it let me install.

yourpepsidrinkwastaken, 2025-07-02:
Right now I tried to install Linuwu-drivers seperately and with make LLVM =1.
This is my current output : 
make LLVM=1
make -C /lib/modules/6.15.4-2-cachyos-bore-lto/build M=/home/yourpepsidrink/DAMX-0.9.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make[2]: Entering directory '/home/yourpepsidrink/DAMX-0.9.1/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 20.1.6
  You are using:           clang version 20.1.7
  CC [M]  src/linuwu_sense.o
src/linuwu_sense.c:4537:9: warning: use of bitwise '&' with boolean operands [-Wbitwise-instead-of-logical]
 4537 |     if (has_cap(ACER_CAP_PREDATOR_SENSE) & !has_cap(ACER_CAP_NITRO_SENSE_V4))
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                          &&
src/linuwu_sense.c:4537:9: note: cast one or both operands to int to silence this warning
src/linuwu_sense.c:4542:9: warning: use of bitwise '&' with boolean operands [-Wbitwise-instead-of-logical]
 4542 |     if (has_cap(ACER_CAP_PREDATOR_SENSE) & has_cap(ACER_CAP_NITRO_SENSE_V4))
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                          &&
src/linuwu_sense.c:4542:9: note: cast one or both operands to int to silence this warning
2 warnings generated.
/bin/sh: line 1: ld.lld: command not found
make[4]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/scripts/Makefile.build:212: src/linuwu_sense.o] Error 127
make[4]: *** Deleting file 'src/linuwu_sense.o'
make[3]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/Makefile:2039: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/home/yourpepsidrink/DAMX-0.9.1/Linuwu-Sense'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.4-2-cachyos-bore-lto/build'
make: *** [Makefile:12: all] Error 2

PXDiv, 2025-07-04:
This might due to be the differences between the compiler. the clang kernel may not support the linuwu-sense drivers. Though i don't know whats going on entirely, i will look into it.
For now i can only suggest you to use some of the more supported distros like base fedora, nobara, arch or ubuntu (officialy tested only on ubuntu)

PXDiv, 2025-07-04:
Please post the same issue in the drivers page [linuwu-sense ](https://github.com/0x7375646F/Linuwu-Sense)

PXDiv, 2025-07-13:
Since there is no update to the last reply in the last one week, I'll assume the issue has been fixed.
Closing the issue now, please reopen the issue if you still have any problems.

closed 2025-07-13 by PXDiv