PXDiv/Div-Acer-Manager-Max #43: Acer Swift sfg14-63 support

SipT1k, 2025-07-10:
It would nice to add support for acer swift go 14 models witch using AcerSence on windows.
The project looks good!

Thank you for your hardwork

PXDiv, 2025-07-10:
What's the current status on you laptop? is it not working?

SipT1k, 2025-07-10:
I use the Nobara distro.
The full name of the laptop is: Acer Swift Go 14 SFG14-63 (NX.KTSEP.003).
The status can be found here: It shows as an unknown model.
[image]

SipT1k, 2025-07-10:
Logs are here:

SipT1k, 2025-07-11:
Tried the same issue: Failed to connect to DAMX daemon. The Daemon may be initializing please wait.

PXDiv, 2025-07-13:
Your model may just not be supported as this was made mostly for nitro and predator laptops.
Swift models may have a whole different set of WMI methods what control the fans and lighing, if you want you can identify all of that fork and modify the linuwu sense to include it instead of the wmi methods included on Predator and Nitro laptops called "AcerGamingControls". It may be called something else in Swift series

PXDiv, 2025-07-13:
There isnt much that we can you here as i dont own any laptops from the swift series, yours looks to be pretty modern, hence i'll assume that it uses WMI methods for its controls, just that they are named different.
I'll Rename this issue to say "Modern Swift Models Not Supported"