PXDiv/Div-Acer-Manager-Max #46: Does this works on Bazzite?

AntonioRocha17, 2025-07-12:
Genuine question from someone who's using linux for the first time and my main goal is gaming.
I'm unable to install Linuwu-Sense and I don't have a clue how to do it

AntonioRocha17, 2025-07-13:
I have a ANV15-51-57WS.
Tried both remote installation and downloaded the file to install

these are the errors I get:

==========================================
    DAMX Remote Installer v1.0.0     
    Acer Laptop WMI Controls for Linux  
==========================================

Checking system compatibility...
✓ systemd found
Kernel version: 6.15.6-101.bazzite.fc42.x86_64
✓ Kernel version 6.15 is supported
Detected OS: Bazzite 42 (FROM Fedora Kinoite)
Note: Only Ubuntu is officially supported. Other distributions may work but are not guaranteed.
System compatibility check completed.
Checking dependencies...
Dependencies check completed.
Fetching latest release information...
Latest release found: DAMX v0.9.1
Download URL: https://github.com/PXDiv/Div-Acer-Manager-Max/releases/download/v0.9.1/DAMX-0.9.1.tar.xz
Downloading DAMX package...
Downloading DAMX-0.9.1.tar.xz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 45.8M  100 45.8M    0     0  11.5M      0  0:00:03  0:00:03 --:--:-- 16.1M
Downloading checksum file...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    84  100    84    0     0    143      0 --:--:-- --:--:-- --:--:--   143
Verifying package integrity...
DAMX-0.9.1.tar.xz: OK
Package integrity verified successfully.
Extracting package...
Package downloaded and extracted successfully.

Starting DAMX Suite installation...
Performing cleanup before installation...
Performing comprehensive cleanup...
Disabling current DAMX-Daemon service...
Removed '/etc/systemd/system/multi-user.target.wants/damx-daemon.service'.
Removing current service file...
Checking for legacy installations...
No legacy installations found.
Removing current installation files...
Comprehensive cleanup completed.

Installing Linuwu-Sense drivers...
make -C /lib/modules/6.15.6-101.bazzite.fc42.x86_64/build M=/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/kernels/6.15.6-101.bazzite.fc42.x86_64'
make[2]: Entering directory '/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense'
make[2]: Leaving directory '/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.15.6-101.bazzite.fc42.x86_64'
make -C /lib/modules/6.15.6-101.bazzite.fc42.x86_64/build M=/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.15.6-101.bazzite.fc42.x86_64'
make[2]: Entering directory '/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense'
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.15.6-101.bazzite.fc42.x86_64'
make -C /lib/modules/6.15.6-101.bazzite.fc42.x86_64/build M=/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.15.6-101.bazzite.fc42.x86_64'
make[2]: Entering directory '/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense'
make[2]: Leaving directory '/tmp/damx-install-7176/DAMX-0.9.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.15.6-101.bazzite.fc42.x86_64'
sudo install -d /lib/modules/6.15.6-101.bazzite.fc42.x86_64/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.15.6-101.bazzite.fc42.x86_64/kernel/drivers/platform/x86
install: cannot create regular file '/lib/modules/6.15.6-101.bazzite.fc42.x86_64/kernel/drivers/platform/x86/linuwu_sense.ko': Read-only file system
make: *** [Makefile:42: install] Error 1
Error: Failed to install Linuwu-Sense drivers
Installing DAMX-Daemon...
Created symlink '/etc/systemd/system/multi-user.target.wants/damx-daemon.service' → '/etc/systemd/system/damx-daemon.service'.
Warning: DAMX-Daemon service may not have started correctly. Check with 'systemctl status damx-daemon.service'
Installing DAMX-GUI...
cp: cannot create regular file '/usr/share/icons/hicolor/256x256/apps/damx.png': Read-only file system
/tmp/setup.sh: line 415: /usr/share/applications/damx.desktop: Read-only file system
DAMX-GUI installed successfully!
Some components failed to install. Please check the errors above.
Installation failed. Please check the errors above.
Cleaning up temporary files...

Sorry I don't know where are the log files, I can upload them if you could tell me the folder they're in.

PXDiv, 2025-07-13:
Please refer to the [FAQ ](https://github.com/PXDiv/Div-Acer-Manager-Max/blob/main/FAQ.md#-driver-installation-fails)for Error 1

PXDiv, 2025-07-13:
the daemon is installed but its somehow not able to start
Try starting it manually by 
`sudo systemctl enable damx-daemon.service`
 `sudo systemctl start damx-daemon.service`
and check with `systemctl status damx-daemon.service`

PXDiv, 2025-07-14:
I dont really have experience with bazzite or immutable linux distros in general.
The only advice i can give right now is to use more tested distros with DAMX like Nobara for gaming and ubuntu or kubuntu for general use.

closed 2025-07-18 by PXDiv