PXDiv/Div-Acer-Manager-Max #47: Acer Nitro AN517-55 Issue

Matixaris, 2025-07-13:
Acer Nitro AN517-55 DAMX loads module is active but laptop type is detected as UNKNOWN and no features are available. this if fixable if needed i can send logs 

[image]

[image]

[image]

PXDiv, 2025-07-13:
Please refer to the [FAQ page](https://github.com/PXDiv/Div-Acer-Manager-Max/blob/main/FAQ.md#-it-shows-unknown-model-and-my-model-isnt-in-the-compatibility-list) as it contains the workaround to your problem, please update me with the results and whats working so that i can update the compatibility list

Matixaris, 2025-07-13:
everything is working thanks

closed 2025-07-13 by Matixaris