PXDiv/Div-Acer-Manager-Max #51: Purge this app

b24fun, 2025-07-17:
How can I get rid of this app with no trace?

NatanAmorim, 2025-07-17:
I'm not the dev of this app, but the installer has a `3) Uninstall DAMX Suite` option:

```
==========================================
       DAMX Suite Installer v0.9.1        
    Acer Laptop WMI Controls for Linux  
==========================================

Please select an option:
  1) Install DAMX Suite (complete)
  2) Install DAMX Suite (without drivers)
  3) Uninstall DAMX Suite
  4) Reinstall/Update DAMX Suite (recommended for upgrades)
  5) Check service status
  q) Quit

Enter your choice [1-5 or q]:
```

PXDiv, 2025-07-18:
You can use the local setup script to uninstall it.
Did you face any issues? Is your model not supported?
You can share your model info so that I can help you with it

b24fun, 2025-07-18:
> You can use the local setup script to uninstall it. Did you face any issues? Is your model not supported? You can share your model info so that I can help you with it

Well kind of, when I activated and deactivated the keyboard 30 secoud delay, the keyboard light stoped working. I fix it but I had to go to the bios and resetting the 30 delay thing (Edit. Maybe that happened because I played with the color of the keyboard even though my laptop don't support it I think). And I want to use as less ram possible, the process that is staying in the background uses ram for nothing. I wish when I open the app to start that process and when I close the app to terminate all the process.

The app is great but it needs some small adjustments for example if your laptop is not compatible with the RGB keyboard to not let you to access the tab and terminate all the process when the app is closed.

XRGC-Digital, 2025-07-26:
How much RAM do you even have to be trying to keep basically any active processes out of memory? Unused RAM is wasted RAM after all.

b24fun, 2025-07-26:
10 MB. I want to make my ram useful, that's why I switch from windows 11 that's uses a whopping 6 GB of ram on fresh startup to Linux that uses 700 MB at startup (According to htop)