PXDiv/Div-Acer-Manager-Max #68: Failed to install Linuwu-Sense drivers Error on Ubuntu - Acer Nitro  ANV15-51.

zeeshan-akram, 2025-08-17:
Hi,
Thank you for this software.
I am facing error while installing this software. My System information is

System: Acer Nitro ANV15-51
OS: Ubuntu 25.04
Kernel version: 6.14.0-27-generic
CPU: Core-I7-13H
GPU: RTX 4060

Please find logs  below for reference, According to my research I have to disable secure boot.
Do I really have to disable secure boot ? or can I re-enable after installing this module?

Logs are-----:
==========================================
    DAMX Remote Installer v1.0.0     
    Acer Laptop WMI Controls for Linux  
==========================================

Checking system compatibility...
✓ systemd found
Kernel version: 6.14.0-27-generic
✓ Kernel version 6.14 is supported
Detected OS: Ubuntu 25.04
✓ Ubuntu detected (officially supported)
System compatibility check completed.
Checking dependencies...
Dependencies check completed.
Fetching latest release information...
Latest release found: DAMX v0.9.1
Download URL: https://github.com/PXDiv/Div-Acer-Manager-Max/releases/download/v0.9.1/DAMX-0.9.1.tar.xz
Downloading DAMX package...
Downloading DAMX-0.9.1.tar.xz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100 45.8M  100 45.8M    0     0  2373k      0  0:00:19  0:00:19 --:--:-- 3238k
Downloading checksum file...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100    84  100    84    0     0     33      0  0:00:02  0:00:02 --:--:--    76
Verifying package integrity...
DAMX-0.9.1.tar.xz: OK
Package integrity verified successfully.
Extracting package...
Package downloaded and extracted successfully.

Starting DAMX Suite installation...
Performing cleanup before installation...
Performing comprehensive cleanup...
Checking for legacy installations...
No legacy installations found.
Removing current installation files...
Comprehensive cleanup completed.

Installing Linuwu-Sense drivers...
make -C /lib/modules/6.14.0-27-generic/build M=/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-27-generic'
make[2]: Entering directory '/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense'
make[2]: Leaving directory '/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0-27-generic'
make -C /lib/modules/6.14.0-27-generic/build M=/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-27-generic'
make[2]: Entering directory '/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
  You are using:           gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0-27-generic'
make -C /lib/modules/6.14.0-27-generic/build M=/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-27-generic'
make[2]: Entering directory '/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
  You are using:           gcc-14 (Ubuntu 14.2.0-19ubuntu2) 14.2.0
make[2]: Leaving directory '/tmp/damx-install-125518/DAMX-0.9.1/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/linux-headers-6.14.0-27-generic'
sudo install -d /lib/modules/6.14.0-27-generic/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.14.0-27-generic/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
modprobe: ERROR: could not insert 'linuwu_sense': Key was rejected by service
make: *** [Makefile:45: install] Error 1
Error: Failed to install Linuwu-Sense drivers
Installing DAMX-Daemon...
Created symlink '/etc/systemd/system/multi-user.target.wants/damx-daemon.service' → '/etc/systemd/system/damx-daemon.service'.
DAMX-Daemon installed and service started successfully!
Installing DAMX-GUI...
DAMX-GUI installed successfully!
Some components failed to install. Please check the errors above.
Installation failed. Please check the errors above.
Cleaning up temporary files...

PXDiv, 2025-08-17:
Did you check the FAQ?
https://github.com/PXDiv/Div-Acer-Manager-Max/blob/main/FAQ.md#-driver-installation-fails

zeeshan-akram, 2025-08-20:
Thanks, @fhenn84-commits . This worked!

I think the creator of Signmod was inspired by f_society from the Mr. Robot show.

By the way, I modified the Signmod repository code to make some good tweaks:

- Added support for encrypted OpenSSL signing keys (previously, it saved the key in plain text on the drive).
- Removed fingerprint information from the certificate (e.g., location and fcodersociety).

If someone need it, I can provide the modified code.